openeew/openeew-sensor #3: BOM and PCB files missing

frankenstein91, 2020-08-12:
As far as I can see, only the circuit diagram is currently available, but the BoM and PCB files are still missing.
But these files are already referred to in the documentation and on the website.

krook, 2020-08-12:
@andygrillo

andygrillo, 2020-08-12:
@frankenstein91 @krook yes sorry we only had a PDF schematic. Please look now- I added the eagle files :)

andygrillo, 2020-08-12:
For now you can generate the BOM from the eagle schematic. I will clean up the BOM I have here and upload later today.

hildogjr, 2020-08-13:
How about using [KiCad](https://kicad-pcb.org/) as free schematic/layout software? There is also [tool for cost BOM](https://pypi.org/project/kicost/) .

andygrillo, 2020-08-13:
Yes I suppose we should stick to open source (although eagle is free for this size of board).

we used eagle because we were comfortable, but if you are ok with doing this board in Kicad, that would be awesome.

chronopoulos, 2020-08-13:
+1 for kicad

gimbas, 2020-08-13:
Would there be interest in a project conversion to Kicad? I want to contribute.

andygrillo, 2020-08-13:
If you are willing to do this (great!), I will list out here my findings from testing the current board that we just got  back from China. The important functions seem good (still testing), i.e. ESP32, UART interface, Ethernet, and most importantly SPI connection with ADXL355.

However would be good to take a look at the WS2812b circuit as I'm not able to get anything other than white right now using the Neopixels sample code, and there is a flicker with the brightness.

The UART header + 5v dc jack should probably be replaced with a USB port (type C). The issue is that some follks may not have a 3.3v FTDI module lying around - will make programming easier.

gimbas, 2020-08-13:
Starting to work on it right away, there are some things that warrant some discussion, where should that happen?

krook, 2020-08-13:
@gimbas The #sensors channel in Slack would probably be the right place.

hildogjr, 2020-08-13:
I would propose to add "manf#" field to each part for the Manufacture Part Number. We can use it by [KiCost](https://pypi.org/project/kicost/) for generate a automatic cost BOM.

andygrillo, 2020-08-29:
> I would propose to add "manf#" or "mpn" field to each part for the Manufacture Part Number. We can use it by [KiCost](https://pypi.org/project/kicost/) for generate a automatic cost BOM.

Thats a great idea @hildogjr  - please can you suggest this in a new issue ?

closed 2020-08-31 by andygrillo