openeew/openeew-sensor #4: Add method for including latitude and longitude of sensor to msgs for /device topic

andygrillo, 2020-08-12:
In previous iterations of the Grillo sensor we included GPS modules. However we have not included them in the sensor hardware to save cost, to make it easier to install away from windows, and because NTP time is accurate enough for EEW.

However we now need a new way to pass on the location of each sensor in the `/device` topic message.

I suggest we avoid using Google Geolocation API as this incurs a cost. Instead we can pass on the lat/lon coordinates using the ESP32 smart config protcol. This is already used to send wifi credentials - why cant it also be used to send lat/lon coordinates of phone ? 

Potentially we can create a new iteration of the ESPTouch mobile app which sends this data using https://pub.dev/packages/esptouch_flutter, and then modify the firmware to receive these new values.

johnwalicki, 2020-11-12:
I found a simple API that will give you Lat,Lng
https://ipstack.com/documentation
I signed up for a free apikey that allows for 10K msgs / month
The API  was simple to call and it returns a json object that includes Lat,Lng
http://api.ipstack.com/check?access_key=<myapikey>

johnwalicki, 2020-11-12:
I envision, at every device powerup, it calls this location API and figures out where it is.
Sends a "Device Activated" message to some restful "Onboarding" interface in the Cloud.
The message includes WiFi, Ethernet mac addresses and Lat,Lng
The response from our centralized "Onboarding" API would include the MQTT host it should connect to.
"Oh, your device is in [Nepal,New Zealand,Mexico,Carribean,etc] return a Watson IoT Platform MQTT instance close to me"

andygrillo, 2020-11-12:
Sounds great, excellent way to route devices to endpoints.

Ill look at this servie tomorrow. My concern is if it works simply by the
ip address. We tries this before and it regularly got the address wrong by
hundreds of meters and in one case in another state. This would have a big
impact on the detection. But maybe we coukd get the user to confirm the
address somehow?

johnwalicki, 2020-11-12:
Agree that it might not be 100% accurate but it gets really close.  In my case, it put the map pin directly in front of my house.

johnwalicki, 2020-11-23:
A mobile application might need to assist in more accurate geolocation.  In the OpenEEW #apps slack channel, we've been discussing using Flutter / SmartConfig / Device registration panels that can send email and location to the cloud so that we can link the device WiFi mac address to an owner / place.

johnwalicki, 2020-12-13:
The mobile OpenEEW Provisioning app sends the smartphone GPS coords as part of the user details registration.
This information is stored in the Cloudant DB.
See https://github.com/openeew/openeew-provisioner/blob/main/PROVISIONING.md

The possible risk is that devices get activated somewhere NOT where they will be positioned.
@andygrillo  is that sufficient?

andygrillo, 2020-12-13:
I suspect this is fine. Later on we could discuss adding a verification
step where the address is sent back to the user, or the user has to confirm
the location on the dashboard ?

johnwalicki, 2020-12-13:
Thinking through the wifi set-up, the device might be activated, using the Mobile app, on one network,  that location would be written to Cloudant db. Then,  when the device moves, the new wifi network said/password will likely be different. The mobile app would be used, call SmartConfig and also adjust/overwrite the device location in the Cloudant db.

Closing issue

closed 2020-12-13 by johnwalicki