openeew/openeew-sensor #6: PoE module plugin

andygrillo, 2020-08-13:
Would be good to have the option to add a PoE module that could be plugged in. This would allow for installation in commercial environments with only an ethernet cable connecting the sensor.

I know its been discontinued, but as an example this seemed to work well :
https://github.com/particle-iot/ethernet-wing

perigoso, 2020-09-06:
I never played with PoE, so I was unfamiliar with it. But I've been doing some research and learning about it so I might be able to implement it in the near future, for now though  I think i detected our first problem with it, our current Ethernet connector is not made with PoE in mind so I think it's not possible to do it with it, we need to look for an alternative.

andygrillo, 2020-09-06:
Good find. Our ethernet circuit is based on an Olimex ESP32 board. I
believe they also have a PoE version so I'll see what components they use
there.

perigoso, 2020-09-06:
Olimex does have some very good reference material, i think this is a good guideline: [ESP32-POE-ISO](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware)

andygrillo, 2020-09-07:
Oh very nice :) They have kicad files also https://github.com/OLIMEX/ESP32-POE-ISO/tree/master/HARDWARE/ESP32-PoE-ISO-Rev.D