openeew/openeew-sensor #7: Send meta data message on boot to /device topic

andygrillo, 2020-08-17:
When the device boots up, it should send a message to `device` MQTT topic containing device ID, latitutde, longitude, firmware version, device type and time.

An example would be:
```{device_id: 2351n2x55, latitude: 19.35, longitude: -99.14, firmware_version: 1.2, device_type: esp32, time_entered: 1597685615}```

This message will become a row in the database - see the corresponding table structure [here](https://github.com/openeew/openeew-detection/blob/master/init_db.sql).

johnwalicki, 2020-11-16:
I've added an OpenEEW Device Activation endpoint to the firmware.  This isn't a MQTT message because the board doesn't know where to send its MQTT data until the endpoint replies with the MQTT broker.   The HTTP post from the board to the endpoint running in IBM Cloud, includes 
```{"macaddress":"112233445566","lat":40,"lng":-74,"firmware_device":"1.0.0"}```

johnwalicki, 2020-11-16:
Not exactly the original request, but very close to the intended purpose.  I think this GitHub issue can be closed via 
https://github.com/openeew/openeew-sensor/pull/43
https://github.com/openeew/openeew-sensor/pull/44
https://github.com/openeew/openeew-sensor/pull/45

johnwalicki, 2020-11-16:
The IBM Cloud side of the new OpenEEW Device Activation endpoint stores the new device details in a Cloudant DB and in Watson IoT Platform metadata.

johnwalicki, 2020-11-23:
Closing issue - implemented via the Device Activation techniques described above.

closed 2020-11-23 by johnwalicki