openeew/openeew-sensor #8: Change RMII pins in Kicad layout

andygrillo, 2020-08-17:
The TXD1 pin from the ESP32 should go to the TXD1 pin in the LAN8720 module.
Similarly, the TXD0 pin should also go to the TXD0 pin in the LAN8720.

Currently we have these crossed:
[image]

Its not possible to reassign these pins in code because the ESP32 has a fixed RMII bus, so the wiring must be done correctly.

perigoso, 2020-08-17:
This was one thing in particular that i wanted to ask when i ported the schematic, makes sense now.

andygrillo, 2020-08-17:
Yes, was our mistake. Its a shame because we wanted to test the LAN8720

perigoso, 2020-08-19:
Check If this is the only erroneous connection in the RMII

commit 3360ea0f (2020-08-19): fix #8
commit 5c874271 (2020-08-19): Merge pull request #9 from perigoso/fix_#8

fix #8
closed 2020-08-19 by andygrillo