openeew/openeew-sensor #12: Rev 1.0 hardware issues

perigoso, 2020-08-29:
> - Ethernet: I think the problem is simply the 2 crossed connectors to RMII bus which I've explained above. Hopefully this is the only issue - the connector is still able to recognize when an ethernet cable is connected or disconnected, it just can't communicate with the router. (edited) 
> 
> - Neopixels: There seems to be a power issue. Its not possible to get all the leds on at same time, and they appear too bright - you can't see the color range. I suspect there is an issue with the voltage difference between the data pin and power pin
> 
> - Buzzer: I can't get sound from this. I've tried many examples such as this https://techtutorialsx.com/2017/07/01/esp32-arduino-controlling-a-buzzer-with-pwm/
> 
> @andygrillo @ Slack sensors

This is issue is meant to keep track of these issues and related fixes

andygrillo, 2020-08-29:
Great idea. Thanks @perigoso this way it doesn't get lost in slack :)

perigoso, 2020-08-31:
I wanted to ask about this circuit? what exactly is it trying to achieve, and where did you get it from?

[image]

perigoso, 2020-08-31:
Changed the level shifting on the WS2812B LEDs to a schmit-trigger buffer, this avoids the inversion of the data line, and should provide clean edges (i believe one these was the cause for the issues).

[image]

perigoso, 2020-08-31:
Changed the Buzzer driving circuit

[image]

@andygrillo If you want to test the buzzer with the current revision try changing the resistor in series with the buzzer for a 0 ohm / short, and change the transistor base resistor for a value around 1k, you should be able to get some sound out of it.

andygrillo, 2020-08-31:
> I wanted to ask about this circuit? what exactly is it trying to achieve, and where did you get it from?
> 

Hi @perigoso , the guy who designed this based the ethernet circuit on the Olimex esp32 board: https://www.olimex.com/Products/IoT/ESP32/ESP32-EVB/  - here you can find the link to the schematic pdf which he worked from.

andygrillo, 2020-09-03:
@perigoso when you receive the board, would you be capable rerouting 2 RMII connections between the LAN8710 and ESP32 ?

I think we need to cut traces between TXD1 > TXD0, and TXD0 > TXD1, then connect with new wires so that TXD1 > TXD1 and TXD0 > TXD0 ?

I have tried but the traces are very small and I don't have a microscrope/my eyes aren't good enough....

[image]

perigoso, 2020-09-03:
@andygrillo I was already planning on doing so! should not be an issue

andygrillo, 2020-09-03:
If you success pls tell me how you did it! its so small I can't even use a normal solid wire between the pads

andygrillo, 2020-09-03:
Also, you'll need an FTDI 3.3v to program it, which I suspect you would already have

gareth2020, 2020-09-11:
We can work with the Plated Though Holes (PTH)s on the underside.
3 tracks are visible on the underside from PTHs under the esp32 to PTHs near the PHY.
We can bare two of those tracks with a scalpel.
We can cut the two tracks with a scalpel and then solder new wires to cross over the connections.

Numbering the holes under the esp32 from top to bottom as h1,h2,h3,
Numbering the holes next to the PHY from top to bottom as p1,p2,p3

On V1.0 pcb
h1 is connected to p1 by a long track
h2 is connected to p2 by a long track
h3 is connected to p3 by a long track

h1 does not matter
h2 seems to be connected to pin 36 of esp32 (IO22 = EMAC_TXD1)
h3 seems to be connected to pin 31 of esp32 (IO19 = EMAC_TXD0)

In the following part of the top view:
p1 is connected to pin 21 of the PHY (TXEN)
p2 is connected to pin  22 of the PHY (TXD0) 
p3 is connected to pin  23 of the PHY (TXD1)
[image]

perigoso, 2020-09-22:
Quick update, after the rework the board seems to work as expected
[image]

```
ETH Disconnected
ETH Connected

ETH MAC: 24:6F:28:CC:E8:A3, IPv4: 192.168.1.69, FULL_DUPLEX, 100Mbps
connecting to google.com
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Tue, 22 Sep 2020 00:07:10 GMT
Expires: Thu, 22 Oct 2020 00:07:10 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
                                                                              <TITLE>301 Moved</TITLE></HEAD><BODY>
                                                                                                                   <H1>301 Moved</H1>
               The document has moved
                                     <A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection
```

andygrillo, 2020-09-22:
great work!

andygrillo, 2020-09-22:
for reference, here is the ethernet test program:
```
/*
    This sketch shows how to configure different external or internal clock sources for the Ethernet PHY
*/
#include <ETH.h>
/* 
   * ETH_CLOCK_GPIO0_IN   - default: external clock from crystal oscillator
   * ETH_CLOCK_GPIO0_OUT  - 50MHz clock from internal APLL output on GPIO0 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO16_OUT - 50MHz clock from internal APLL output on GPIO16 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO17_OUT - 50MHz clock from internal APLL inverted output on GPIO17 - tested with LAN8720
*/
#ifdef ETH_CLK_MODE
#undef ETH_CLK_MODE
#endif
#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT
//#define ETH_CLK_MODE ETH_CLOCK_GPIO0_IN
//#define ETH_CLK_MODE ETH_CLOCK_GPIO0_OUT
// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   -1
// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720
// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        0
// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23
// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18
static bool eth_connected = false;
void WiFiEvent(WiFiEvent_t event) {
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}
void testClient(const char * host, uint16_t port) {
  Serial.print("\nconnecting to ");
  Serial.println(host);
  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }
  Serial.println("closing connection\n");
  client.stop();
}
void setup() {
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
}
void loop() {
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}```

perigoso, 2020-09-22:
I got the buzzer to output sound, i have no way to measure but it's not too loud, some optimizations can be done.

[I posted a video on slack](https://openeew.slack.com/archives/C014ZP4AM4H/p1600812352049000)

The test code for reference:
```
int freq = 4000;
int channel = 0;
int resolution = 8;
int io = 32;
  
void setup() {
  ledcSetup(channel, freq, resolution);
  ledcAttachPin(io, channel);

  ledcWriteTone(channel, freq);
  ledcWrite(channel, 125);
}
  
void loop() {

}

```

perigoso, 2020-09-22:
Leds seem to work as desired after the proposed changes.

[video on slack](https://openeew.slack.com/files/U019H72A2LQ/F01B0756AB0/vid_20200923_004521_ls.mp4)

the rework:
[image]

the test code:
```
#include <Adafruit_NeoPixel.h>

#define LED_PIN 16
#define LED_COUNT 3

//Adafruit_NeoPixel pixels = Adafruit_NeoPixel(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)

void setup()
{
//pixels.begin(); // This initializes the NeoPixel library.
  strip.begin();
  strip.show(); // Initialize all pixels to 'off'
}

void loop()
{
    for (int i = 0; i < 2; i++) {
    for (int j = 0; j < 2; j++) {
    for (int k = 0; k < 2; k++) {
    for (int l = 0; l < 3; l++) {
        strip.setPixelColor(l, strip.Color(i * 255, j * 255, k * 255)); // Moderately bright green color.
        strip.show(); // This sends the updated pixel color to the hardware.
        delay(200); // Delay for a period of time (in milliseconds).
    }}}}
}
```

closed 2020-09-23 by perigoso