openeew/openeew-sensor #16: Add alarm function

andygrillo, 2020-09-03:
Add MQTT subscribe function that interrupts the main loop when an alarm message is received.

johnwalicki, 2020-09-30:
I'm looking at implementing this alarm topic in the OpenEEW firmware / Watson IoT Platform implementation

johnwalicki, 2020-11-26:
- Andy posted a sample sketch in the firmware slack channel today that exercises the buzzer on the production board.
- Add that example into the Alarm topic callback so that we can send Alarm alerts from the cloud to the devices and warn of impending quakes. 
- Build a sample Node-RED flow that publishes to the Alarm topic

johnwalicki, 2020-12-20:
version130 branch has a very simplistic alarm function that sounds the buzzer when the ESP32 receives an earthquake MQTT topic.
https://github.com/openeew/openeew-sensor/commit/6928ee0ae6397e37abbbe023f37046f5d725f54a

johnwalicki, 2021-02-19:
Firmware v1.3.0 has been published
This Issue has been closed via https://github.com/openeew/openeew-sensor/pull/61

closed 2021-02-19 by johnwalicki