openeew/openeew-sensor #17: Missing bypass capacitors on the accelerometer chip power supply and voltage reference pins.

gareth2020, 2020-09-08:
The noise performance might be worse than the specification because of missing bypass capacitors on the accelerometer chip power supply and voltage reference pins.

Figure 59 in https://www.analog.com/media/en/technical-documentation/data-sheets/adxl354_adxl355.pdf  suggests 2 bypass capacitors should be used for each power supply near to the ADXL355. i.e. 0ne 0.1uF (100nF) capacitor and one 1 uF capacitor.  However the OpenEEW schematic only seems to show 100nF capacitors connected to pins 5,8,10 and 11 of U3.

Analog Devices, probably characterized the noise spectral density for ±2 g range as 22.5 μg/√Hz
by using the circuit shown in Figure 59 of https://www.analog.com/media/en/technical-documentation/data-sheets/adxl354_adxl355.pdf. If the 1u F capacitors are not present, then the noise spectral density might be worse and the signal might be less immune to power supply noise e.g. from other electrical activity on the pcb and from radio frequency interference and from noise on the 5V power input.

For more on why two capacitors per voltage pin see https://www.analog.com/media/en/training-seminars/tutorials/MT-101.pdf

Reliability of the OpenEEW sensor might also be improved with two capacitors per voltage pin because when one of the capacitors fails it might not causing too much damage to the noise performance. If there was only one capacitor then the noise performance might be damaged too much if it failed.

gareth2020, 2020-09-08:
It is hard to tell if the loss of noise performance and immunity will matter in this application. It might not because most local noise will not correlate between sites.  

It is also interesting to note that the Analog Devices evaluation board has two capacitors per voltage pin but they are not different.  See https://wiki.analog.com/_media/resources/eval/user-guides/301-736_adxl355z_evaluation_board_rev_2_-_schematic.pdf

gareth2020, 2020-09-08:
There is another eval board https://wiki.analog.com/_media/resources/eval/user-guides/eval-adicup360/hardware/adxl355/02_043217b_top.pdf   which has 10uF and a 0.1uF connected to each supply pin:
[image]

gareth2020, 2020-09-08:
We could see if adding a 1uF capacitor or a 10 uF capacitor has any affect on the noise amplitude in the /sensor-traces?orgId=2&refresh=5s

perigoso, 2020-09-23:
Reopened, there are still some tests that can be done

andygrillo, 2020-09-23:
@perigoso you mean testing the board as we have it now without bypass capacitors, against a revision with the bypass capacitors ?

perigoso, 2020-09-23:
there was mention of experimenting with 1uF and 10uF, that's what i meant.
though imo there isn't really a need for this

gareth2020, 2020-09-24:
The analog supply to the ADXL355 needs a ferrite in the analog return path as well to create a clean AGND .  AGND is also where the capacitors connected to V1PANA and VSUPPLY should be connected. See https://wiki.analog.com/_media/resources/eval/user-guides/eval-adicup360/hardware/adxl355/02_043217b_top.pdf
Layout advice is given in https://www.analog.com/media/en/training-seminars/tutorials/MT-101.pdf 

I have the two EVAL-ADXL355-PMDZ boards if looking at it will help.

perigoso, 2020-10-26:
I think we can close this, If we end up doing further testing or find issues we can take a further look, but for now this should be resolved.

PR #24 (2020-09-23): Fix #17, accelerometer power
Closes #17 

before:
[image]

after:
[image]
commit 2de6efff (2020-09-23): Merge pull request #24 from perigoso/fix_#17

Fix #17, accelerometer power
commit 8b168617 (2020-09-23): fix: #17, accelerometer power
closed 2020-10-28 by perigoso