openeew/openeew-sensor #52: Program RGB Leds for debugging

andygrillo, 2020-12-04:
This is a discussion for what the LEDs could indicate on the sensor board, to help users understand the status of the device.

- [ ] breathing CYAN: searching for WiFi
- [ ] PURPLE: can't find Wifi
- [ ] blinking BLUE: everytime a msg is sent to cloud (1s)
- [ ] PURPLE: can't find Wifi
- [ ] PURPLE: can't find Wifi
- [ ] PURPLE: can't find Wifi

johnwalicki, 2020-12-04:
In this initial [Blink NeoPixel LED Status stages commit](https://github.com/openeew/openeew-sensor/commit/9c455e9434fc359a945d849a01808d64c77beaf4#diff-c3bf47ea1f4a4a605470313cacb3a44f4a461f68c6faeab07e737610cb5ac835) from several weeks ago, I emulated the Particle Photon color indicators from
https://docs.particle.io/tutorials/device-os/led/photon/

johnwalicki, 2020-12-04:
I like your LED color proposals.

johnwalicki, 2021-02-19:
Firmware v1.3.0 has been published
This Issue has been closed via https://github.com/openeew/openeew-sensor/pull/61

closed 2021-02-19 by johnwalicki