openeew/openeew-sensor #55: resync time on long running sensors

johnwalicki, 2020-12-10:
Sensors might be running for many weeks / months and the time on the ESP32 might drift.
At power up, the ESP32 syncs its time.  That allows it to connect to MQTT TLS.
That's great but weeks later the time might have drifted.
We should implement a periodic NTP sync.  Maybe once a day?
There already is a SetTimeESP32() function that would be called.
Implementation would be an unsigned integer counter in the loop() that just counts up to 86400 and calls the SetTimeESP32() function.

andygrillo, 2020-12-10:
Yes great idea. Could it be more frequent, say every hour? Drift over one
day could be 10s or more...

johnwalicki, 2021-03-30:
Closing this ticket via 
https://github.com/openeew/openeew-firmware/issues/1

closed 2021-03-30 by johnwalicki