openeew/openeew-sensor #57: Heartbeat MQTT message

johnwalicki, 2020-12-10:
Today, the sensor sends a MQTT message every second that contains accelerometer data.
If we're getting data, we know the board is alive.
If we implement an edge based STA / LTA algorithm, the sensor will process the accelerometer data at the edge.  If there are no earthquakes, we might not hear from a board for weeks/months.  We won't know its alive.

When we implement STA/LTA edge processing, we should also implement a heartbeat MQTT topic to replace the MQTT data. It does not need to be every second.   The MQTT message would only need to send 1 byte of data.  We can derive the macaddress from the device id (part of the Watson IoT headers).  In the cloud, we could either update the Cloudant record with a last heartbeat timestamp.  The OpenEEW dashboard could read that to indicate status.
There's a Watson IoT Last Event rest api but the dashboard would need to know those credentials.  I could find an example of that implementation too.

Maybe once a minute -  would be 1440 messages/day
Maybe once a hour - would be 24 messages/day

If we're worried about the IBM Cloud Watson IoT Platform lite tier, its limited to 200MB/month so we would want to keep our heartbeats small / sufficiently infrequent.

andygrillo, 2020-12-10:
Doesnt the MQTT broker automatically track devices that are connected to it
? isnt there an inbuilt heartbeat that is used by MQTT which we can simply
read from node-red ?

johnwalicki, 2020-12-10:
Yes, Possibly.  There is a Node-RED node to query the status of a device.   I have a flow which can enumerate all of the devices.  It then can iterate across all the devices and get status for each one.  I've never used that technique as a heartbeat.  Will need to experiment to see if I can discover if they are up/down

I see:
payload.status.alert.enabled is either true or false
payload.status.alert.timestamp

johnwalicki, 2020-12-11:
There are other techniques to get the MQTT status of a device.  Will implement this for the OpenEEW Dashboard in a different way.  Saves sending useless messages to the cloud. Closing ticket.

closed 2020-12-11 by johnwalicki