openeew/openeew-sensor #58: Schematic: custom CP2102N schematic symbol is not included into Sensor_Motion_local.lib

hovercraft-github, 2021-01-16:
The current schematic (both master and version130 branches) uses custom (or outdated) symbol CP2102N-A01-GQFN24  
for the U6 part, and this custom symbol is not included into Sensor_Motion_local.lib file, so the schematics diagram looks unreadable in every KiCad instance using an up-to-date schematic symbol libraries.  
Although the official KiCad library provides an equivalent schematic symbol in the current repository, it can not be directly substituted because of different geometry.  

Please do one of the following:

- put custom CP2102N-A01-GQFN24 symbol into Sensor_Motion_local.lib file
- or (preferably) replace the custom CP2102N-A01-GQFN24 symbol with an equivalent one from the current KiCad repository

perigoso, 2021-01-20:
I'm sorry but I was not able to reproduce your issue.

The symbol is linked to the official Kicad libraries, and has not been modified to my knowledge.

As of version 5.1.9 of Kicad, it's latest stable release, the symbol seems to be linked correctly.

@andygrillo can you take a look at the schematic? It should not matter but I am on a different operating system than windows.

@hovercraft-github can you provide more detail? what operative system, what version of kicad and symbol library, etc.

hovercraft-github, 2021-01-24:
@perigoso Thank you for your attention and support!
Your are right in that the symbol your are using in your schematic is linked to the official Kicad libraries shipped with the version 5.1.9 of Kicad.
But Kicad symbol libraries itself has separate github repository.
It is common practice to update these libraries periodically while they maintain compatibility with the current stable Kicad release,
because of updated and new components they provide.
It looks like that after release of version 5.1.9, the Kicad library maintainers crew decided to remove the CP2102N-A01-GQFN24 symbol from their repository and replace it with an equivalent one with different geometry.
From this on, the problem I described come into play:
anybody who updates his libraries from the Kicad library repository, will get unreadable representation of the CP2102N symbol in openeew-sensor schematic.
At least it happened to me. :)
So, basically it is not an openeew-sensor issue but rather Kicad libraries incompatibility.
Maybe, it would make sense to denote it somewhere in the documentation.

perigoso, 2021-01-24:
This is intended behavior, the master branch of the libraries is not stable, I'm also a contributor there, we try to always be using the latest **stable** release of Kicad and libraries.
You can have both the master and the stable release in parallel in your system fyi.

You're right that this should be documented, could you please open an issue for that?

perigoso, 2021-01-24:
Sorry, don't open an issue, this one is enough, just needs a clearer name