openeew/openeew-sensor #59: Create firmware variant for standard MQTT broker

andygrillo, 2021-01-19:
To allow for non-Watson IoT connections (eg. Mosquitto MQTT), it would be good to have a variant of the Watson IoT (or better yet a configuration option) that allows it to connect to a generic MQTT broker.

johnwalicki, 2021-02-22:
To run an MQTT broker on your local subnet (for instance on a Raspberry Pi), I have made the minimum number of changes to the production firmware via this patch.  Essentially its 3 #ifdefs
Toggle #define LOCALNET in config.h

There's a patch to the version140 branch:
Run a MQTT Broker on a local subnet
https://github.com/openeew/openeew-sensor/commit/df6158a5a825ed4111e3dbc4cf5d02b29ca863e5

johnwalicki, 2021-02-22:
Thanks to @andygrillo  for some inspiration

closed 2021-02-22 by johnwalicki