openeew/openeew-sensor #60: Accelerometer data to Cloud - Continuous Send vs STA/LTA

johnwalicki, 2021-02-07:
The v1.30 firmware runs silently in STA/LTA mode, just holding a 10 second FIFO queue.
It also subscribes to a MQTT_TOPIC_SENDACCEL topic
```iot-2/cmd/sendacceldata/fmt/json```
which, if published from the Cloud/Dashboard, sends a json object to the sensor:
```
{"LiveDataDuration":<integer>}
```
The device parses this MQTT message: 
` numSecsOfAccelReadings = cmdData["LiveDataDuration"].as<int32_t>();`
The device then starts counting down and sending live accelometer data to the cloud. The payload is n seconds of data

Some regional networks / scientists might want a continuous stream of accelerometer data.
This obvious requires more bandwidth and will generate lots of MQTT data.  Someone will have to pay but we can implement the feature.

To keep the code modifications easy, I suggest changing this to an unsigned integer 32 `uint32_t`
That is 4294967295 seconds (which is 136 years or close to 50000 days).  That's sufficiently large!
There would be no other changes to the firmware required.

johnwalicki, 2021-02-07:
@andygrillo Would that be sufficient?

johnwalicki, 2021-02-07:
@vkuna Would that be sufficient?

vkuna, 2021-02-07:
I don't have a strong opinion. If this is an easy solution and it would be easy to change the value remotely, I think it should work just fine.

johnwalicki, 2021-02-07:
Here's how I imagine this working:
- the device checks in when it powers up
- The device mgmt endpoint determines which network this device belongs to and replies to the board with the MQTT broker it should data to.  The board subscribes to a variety of topics (including the sendacceldata topic)
- The device mgmt endpoint also could check a parameter in the device registry (Cloudant) that determines which mode this sensor should operate in (continuous vs STA/LTA)
- The device mgmt endpoint publishes the above message to the device - telling it stream data for max uint seconds.

We might have to worry about timing? What happens to published messages before the device subscribes to the topic?

johnwalicki, 2021-02-07:
Or...
We could create another MQTT topic that "declares" the algorithm that the device should use.
Today, the device firmware implements STA/LTA (local 10 second algorithm) or blindly sending all the accel readings.
Maybe in the future the firmware will implement other seismic algorithms?
We could future proof the firmware so that this potential MQTT topic could tell the device to switch between algorithms.

johnwalicki, 2021-02-07:
At the implementation level for "continuous send mode" it just sets `numSecsOfAccelReadings = 4294967295` seconds. 
We would then be able to extend the firmware with whatever else algorithm a seismologist invents.

vkuna, 2021-02-07:
Yes if we can keep that option open it would be great. STA/LTA is a standard and reliable option, however, we want to try some ML detectors deployed on sensors in future. Plus there might be more changes to station functionality, like allowing station to issue 'on-site' warning etc.. All of these would need significant changes to the firmware and we should be able to upgrade all sensors remotely.

andygrillo, 2021-02-08:
I like this 2nd option better because it means the algorithm change can be
done at any time, not only when boot up. \

If I understand correctly, this would allow the change to be done on the
fly via the dashboard.

johnwalicki, 2021-02-08:
Without any changes to the firmware, I implemented the following UI change in the OpenEEW Device Mgmt interface:
http://openeew-devicemgmt.mybluemix.net/ui/#!/1

Essentially, its sending the maximum int32_t which is 2147483647 seconds ( ~ 68 years )

andygrillo, 2021-02-08:
looks great!
[image: image.png]

johnwalicki, 2021-02-08:
The idea to dynamically change the algorithm is interesting but potentially out of scope for this release.  We currently only have one algorithm.

andygrillo, 2021-02-08:
Sorry my bad description, I just meant change from continuous to triggered,
which it seems you have achieved.

johnwalicki, 2021-02-19:
Firmware v1.3.0 is released.
This issue is closed via https://github.com/openeew/openeew-sensor/pull/61

closed 2021-02-19 by johnwalicki