openeew/openeew-sensor #65: Stops running after reprogramming

neilh10, 2021-06-01:
A device A8032A4DD650 used on a testBoard and is screwed down : after reprogramming it runs successfully.  

Works first time after programming,(flash-openeew.sh) 
[flash-openeew.sh.txt](https://github.com/openeew/openeew-sensor/files/6577682/flash-openeew.sh.txt)

but after a power cycle stops jumping to main, just cycles.

[2021-05-26 15:05:24] rst:0x10 (RTCWDT_RTC_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
[2021-05-26 15:05:24] configsip: 0, SPIWP:0xee
[2021-05-26 15:05:24] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[2021-05-26 15:05:24] mode:DIO, clock div:2
[2021-05-26 15:05:24] load:0x3fff0018,len:4
[2021-05-26 15:05:24] load:0x3fff001c,len:1044
[2021-05-26 15:05:24] load:0x40078000,len:10124
[2021-05-26 15:05:24] load:0x40080400,len:5828
[2021-05-26 15:05:24] csum err:0x6c!=0x4d
[2021-05-26 15:05:24] ets_main.c 371 
[2021-05-26 15:05:24] ets Jun  8 2016 00:22:57
[2021-05-26 15:05:24] 
[2021-05-26 15:05:24] rst:0x10 (RTCWDT_RTC_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
[2021-05-26 15:05:24] configsip: 0, SPIWP:0xee
[2021-05-26 15:05:24] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[2021-05-26 15:05:24] mode:DIO, clock div:2
[2021-05-26 15:05:24] load:0x3fff0018,len:4
[2021-05-26 15:05:24] load:0x3fff001c,len:1044
[2021-05-26 15:05:24] load:0x40078000,len:10124
[2021-05-26 15:05:24] load:0x40080400,len:5828
[2021-05-26 15:05:24] csum err:0x4e!=0x4d
[2021-05-26 15:05:24] ets_main.c 371 
[2021-05-26 15:05:24] ets Jun  8 2016 00:22:57

left running, 5hours later still not jumped

[2021-05-26 20:36:30] rst:0x10 (RTCWDT_RTC_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
[2021-05-26 20:36:30] configsip: 0, SPIWP:0xee
[2021-05-26 20:36:30] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[2021-05-26 20:36:30] mode:DIO, clock div:2
[2021-05-26 20:36:30] load:0x3fff0018,len:4
[2021-05-26 20:36:30] load:0x3fff001c,len:1044
[2021-05-26 20:36:30] load:0x40078000,len:10124
[2021-05-26 20:36:30] load:0x40080400,len:5828
[2021-05-26 20:36:30] csum err:0x6d!=0x4d
[2021-05-26 20:36:30] ets_main.c 371 

Powered off, tried again a couple of days later

[2021-05-28 11:49:24] rst:0x10 (RTCWDT_RTC_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
[2021-05-28 11:49:24] configsip: 0, SPIWP:0xee
[2021-05-28 11:49:24] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[2021-05-28 11:49:24] mode:DIO, clock div:2
[2021-05-28 11:49:24] load:0x3fff0018,len:4
[2021-05-28 11:49:24] load:0x3fff001c,len:1044
[2021-05-28 11:49:24] load:0x40078000,len:10124
[2021-05-28 11:49:24] load:0x40080400,len:5828
[2021-05-28 11:49:24] csum err:0x49!=0x4d
[2021-05-28 11:49:24] ets_main.c 371 
[2021-05-28 11:49:25] ets Jun  8 2016 00:22:57
[2021-05-28 11:49:25] 
[2021-05-28 11:49:25] rst:0x10 (RTCWDT_RTC_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
[2021-05-28 11:49:25] configsip: 0, SPIWP:0xee
[2021-05-28 11:49:25] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[2021-05-28 11:49:25] mode:DIO, clock div:2
[2021-05-28 11:49:25] load:0x3fff0018,len:4
[2021-05-28 11:49:25] load:0x3fff001c,len:1044
[2021-05-28 11:49:25] load:0x40078000,len:10124
[2021-05-28 11:49:25] load:0x40080400,len:5828
[2021-05-28 11:49:25] csum err:0x69!=0x4d
[2021-05-28 11:49:25] ets_main.c 371 
[2021-05-28 11:49:25] ets Jun  8 2016 00:22:57
 

[2021-05-28 11:49:24] ets Jun  8 2016 00:22:57

johnwalicki, 2021-06-01:
Hi Neil - This ticket is potentially opened in the wrong repo.  OpenEEW firmware is discussed over in the 
https://github.com/openeew/openeew-firmware
repository.

I suggest that you try to flash the board again.  Its not booting at all. Grab the binaries from here:
https://github.com/openeew/openeew-firmware/releases/download/1.5.0/openeew-firmware150.zip

neilh10, 2021-06-01:
Hi John. I put it here because it seems like it is at base board support level or at an (embedded?) flash. 
I flashed it three times, (edited above) with openeew-firmware151 and it works until it power cycles. 
The other two boards are good with flashing openeew-firmware151.

So are you suggesting regressing back to openeew-firmware150 to see if that makes a difference?

johnwalicki, 2021-06-01:
I wasn't advocating for v1.5.0 vs v1.5.1 binaries.  It was just the easiest link I could send you.
Would be easier to watch the serial terminal and diagnose what's happening to your boards.
If you think you can share them via a WebEx, I'll send you a link.

neilh10, 2021-06-01:
I have it setup connected through a powered USB Hub to  a riPi4/Ubuntu - so could try WebEx.

neilh10, 2021-06-01:
I was able to reprogram it with openeew-firmware151, with John watching/cell, add it to the network and it booted.
It then was able to access WiFi, and also 1G network, but not my 8 Port 10/100Mbit router.
I rebooted a couple of times, and it came up every time. John reset it remotely a couple of times and it came up every time.
Closing - as this device A8032A4DD650  is now reprogrammed OK.

closed 2021-06-01 by neilh10