openeew/openeew-sensor #66: Doesn't work with all ethernet routers

neilh10, 2021-06-01:
Two openEEW sensors I have only work with some ethernet routers

Device AB032A4DCCCC I use in my office  ,
 works OK with D-link 1Gigbit 16 channel
problems with D-link 10/100M 16Channel and other  8-port 10/100 Mbit routers

Device A8032A4DCCB4 (testBoard screwed down): 
works with tplink AC1750 Wireless Dual Band Gigabit Router
non working ethernet against my standard workshop 18-port 10/100 Mbit  router that works with the above tplink AC1750 and rPi devices.
[image]

https://www.microchip.com/wwwproducts/en/LAN8710A says it works with 10BASE-T/100BASE-TX transceivers

johnwalicki, 2021-06-01:
Hi Neil - This ticket is potentially opened in the wrong repo. OpenEEW firmware is discussed over in the
https://github.com/openeew/openeew-firmware
repository.

For certain, your boards are experiencing odd ethernet behavior.

I've only tested my boards with a TP-Link 1GB ethernet switch.

The firmware is using the standard Arduino / Espressif ESP32 ethernet library.  Its not doing anything special to it.
~/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/ETH.h

johnwalicki, 2021-06-01:
Looking for a pattern in your ethernet switch / router network configurations, it seems the board and ethernet is working with relatively modern 1Gb routers / switches but fails with older 10/100 switches.

I'll dig around for an old 10/100 ethernet switch to test.

johnwalicki, 2021-06-01:
I plugged my (new) OpenEEW device into a DLink 10/100 Ethernet switch:
```
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
ETH Disconnected
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
ETH Connected
Waiting for Ethernet to start...
Waiting for Ethernet to start...
```

neilh10, 2021-06-01:
Hi John, I put it here because its not clear if its a hardware issue or a firmware issue. 
Looks like you are getting the same response .....  good to duplicate.... 
As the https://www.microchip.com/wwwproducts/en/LAN8710A  says it should work with 10/100 - but its actually only worked for me on 1G routers . Possibly suggests that the 1G routers are compensating for something not working.
Practically  Arduino / Espressif ESP32  is a "teaching" principles - not drowning in the historical detail.  
https://en.wikipedia.org/wiki/IEEE_802.11

Traditionally, for a commercial product, the process is define all the IEEE 802.11 standards it needs to work to, and then probably buy some expensive test equipment that will run a regression test.  

This is a bit of statistical method  ~ a few pieces of equipment that work with other equipment.

I can move it to firmware if you think there is a firmware solution.

johnwalicki, 2021-06-01:
Neil and I ran some live experiments.  I was able to find 4 old 10/100 Ethernet switches.  On two (identical models) DLink 10/100 DS-55+ the ESP32 device would just loop forever waiting for the Ethernet to get an IP address.  Two other devices (NetGear 8 port 10/100 FS108) and a DLink DI-524 10/100 router, the ESP32 got an Ethernet address and proceeded through its initialization correctly.

For certain, there is a problem with some old 10/100 Ethernet switch hardware.

It does point out an unlikely scenario / bug in the firmware.  In v1.5.1 or lower, it gets stuck in a while() loop waiting for an Ethernet address.  I've created a firmware patch that will only wait 10 seconds for an IP Address. If it doesn't get an IP address on the Ethernet interface, it exits the while loop and proceeds through the rest of the initialization .

Here's a snip of the behavior with the new patch described above:
```
OpenEEW Sensor Application
ETH Started
ETH Connected
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
Waiting for Ethernet to start...
ESP32 WiFi interface ready
Stored networks : 1
ESP32 WiFi started
Completed scan for access points
WiFi Network scan done
8 network(s) found
1: wyndwpa (-23) 
2: wynd565o (-56)*
3: 565guest (-56)*
ESP32 WiFi interface ready
Reading stored networks from NVM
ESP32 WiFi started
ESP32 WiFi started
```

perigoso, 2022-03-03:
Have any of you tried with different cables? A straight through cable would be the correct one i believe, but do try with a crossover too, nowadays hardware deals with the crossover automatically but this wasn't always the case.

johnwalicki, 2022-03-03:
We eventually agreed that there is a problem with some old 10/100 Ethernet switch hardware.    I tried multiple routers. See tests above.
Closing issue.

closed 2022-03-03 by johnwalicki