openeew/openeew-sensor #69: broken URL in readme

sknebel, 2022-04-03:
The link to the blog post in the first sentence of the readme is 404: https://openeew.com/blog/sensor-benchmark, and from a quick look over the site I couldn't find a new URL.

sergeant82d, 2022-04-03:
It looks to me like the whole project is stalled. I'd love to get a sensor and get it online, but if the network is not working... What's the point?

andygrillo, 2022-04-04:
Hi @sergeant82d , yes we (grillo) are developing behind the scenes some updates including a neural network detection system. Soon this will be published here.

FYI for sometime OpenEEW had stopped being a global network for anyone to plug in to. We had decided as a community that instead we would provide folks the tools to host their own EEW network. This is because we realized a global network would never achieve the density necessary.

sergeant82d, 2022-04-04:
Ahh. Okay, that makes sense. I looked for the information on setting up a network once, but it was missing. Is the local network setup information available now? Any idea if or when the assembled units will be available again? Thanks!